espressif/idf-extra-components #513: feat(esp_encrypted_img): adds support for ECIES HMAC seeds generation within the device @ boot sequence

g-380, 2025-06-16:
### Is your feature request related to a problem?

Hello,

The current design assume that the device hmac key must be securely provisioned onto the device by burning it into an eFuse key block, before it is later used by compute_ecc_key_with_hmac(..) to get the private key ready for use.

To resolve the best secure path for the HMAC key not being exposed during injection, you may consider having the **device generating the hmac key itself** when bootstraping, after application has been verified (as an example)
Could be using IDF random librairies by enabling an entropy source for RNG, if RF subsystem is disabled ?

Thanks for your support

Ashish285, 2025-06-17:
Hi @g-380,

Thank you for your thoughtful suggestion regarding generating the HMAC key on-device during bootstrapping.

While generating the HMAC key using on-device entropy sources (e.g., `esp_random()`) after firmware verification is technically feasible — especially with Secure Boot enabled to ensure code authenticity — it introduces a key architectural challenge for the OTA system.

The current pre-encrypted OTA design assumes that all devices share a common decryption key (or derive it in a deterministic and consistent way). This allows the OTA server to host a single encrypted firmware image that can be safely downloaded and decrypted by any device in the field.

If each device were to generate its own unique HMAC key (and thereby its own ECC private key), this would result in:

* A unique decryption key per device, requiring the OTA server to individually encrypt firmware for each device.
* A mechanism to register or transmit each device's public key back to the OTA infrastructure.
* Significantly increased complexity in OTA server operations, image generation, and device-key association.

Since Espressif typically does not manage the OTA servers in customer deployments, we have no reliable way to ensure such a per-device key management pipeline exists or can scale. Therefore, the pre-encrypted OTA design intentionally opts for a shared HMAC key, securely provisioned to the device (e.g., via eFuse) during manufacturing.

That said, we do recommend that:

* This shared key be stored in a non-readable eFuse block, and
* If needed, partition devices into security groups or batches, each with its own shared key, to improve isolation without sacrificing OTA scalability.

Your proposal is certainly valid in environments where per-device provisioning and OTA customization are feasible (e.g., enterprise deployments or secure provisioning factories). But for general-purpose and scalable OTA updates, a shared or group-shared key model remains the most practical balance between security and deployment efficiency.

Please let us know if you have any specific requirements or use case.

Thanks

Ashish285, 2025-06-19:
Hi @g-380,

We will add the TRNG with SAR ADC based entropy source to the official documentation and also provide an API to extract the public key. 

Let me know if you would like anything else to help support your use case.

Thanks

g-380, 2025-06-19:
Brillant, thanks !

Ashish285, 2025-07-11:
Hi @g-380 ,

We have introduced API to extract public key along with the documentation update with [v2.5.0](https://github.com/espressif/idf-extra-components/tree/master/esp_encrypted_img). I believe this will be enough for the per device unique key use case.

I am closing this issue, please feel free to try it out and create another issue if you encounter any problems.
Thanks

closed 2025-07-11 by Ashish285

Ashish285, 2025-10-15:
Hi @g-380,

Thanks for the question. The `esp_encrypted_img_pbkdf2_hmac_sha256` is not an overkill in this scenario.

You are correct about the entropy, it does come from the HOST `(esp_enc_img_gen.py / generate_hmac_key )` while generating the HMAC key. The `pbkdf2` operation doesn't add to the entropy but helps how this randomly stored key is used on the device

1. Hardware HMAC Utilization: The `esp_hmac_calculate()` function uses ESP32's HMAC peripheral, ensuring the key in eFuse never appears in software-accessible memory. This means the only time derived key is exposed in the application is when it is required. This provides **Key Separation**
2. Side Channel Analysis: Although a single HMAC operation might also be cryptographically sufficient, doing multiple iterations may help increase the complexity required for such analysis.

While you're correct that in the current implementation, `esp_encrypted_img_pbkdf2_hmac_sha256` doesn't add randomness (since the PBKDF2 salt is hardcoded in firmware), the scheme architecture actually supports a more secure approach.

With the current scheme all devices derive the same ECC key from the same HMAC key, which limits the security benefits to side-channel attack resistance rather than cryptographic diversity.

But if you have control over the server infrastructure (beyond simple file hosting), the protocol already provides the capability to implement per-device or per-image random PBKDF2 salts:

* Server generates a random salt for each encrypted image
* Salt is included in the encrypted image header (using reserved space)
* Device extracts the salt from the header during decryption
* Device uses the random salt with `esp_encrypted_img_pbkdf2_hmac_sha256` to derive a unique ECC key

This would provide:

* Per-device key diversity - Each device/image combination derives a different ECC key
* Forward secrecy - Compromising one image doesn't compromise others
* Key rotation - Without re-provisioning devices, new images use new derived keys

We designed the current implementation to support simple file hosting servers without additional capabilities (e.g., static HTTP/S3 servers). These servers cannot:

* Generate device-specific encrypted images
* Maintain per-device state
* Perform cryptographic operations

For such deployments, a hardcoded salt is a pragmatic choice that still provides essential side-channel attack resistance through the 2048 PBKDF2 iterations.

Please let me know if this answers your question. Thanks!