espressif/idf-extra-components #523: [esp-nghttp]: ability to tune NGHTTP2_INBOUND_BUFFER_LENGTH

bryghtlabs-richard, 2025-07-02:
### Is your feature request related to a problem?

I don't have internal space for more threads, and there's plenty of buffering at TCP & TLS layers for my application already.

### Describe the solution you'd like.

I'd like the ability to adjust NGHTTP2_INBOUND_BUFFER_LENGTH through sdkconfig.

bryghtlabs-richard, 2025-08-28:
> Probably working with upstream maintainers will help.

Is there anything else I should do besides the pull request?

> I think changes may go beyond https://github.com/nghttp2/nghttp2/pull/2470, as I can see few more places where there is assumption about max inbound buffer length.

I double-checked our fork and upstream but I only saw it used in `nghttp2_session_recv()`. I believe all the uses of said buffer involve involve `sizeof(buf)`. We've been using that patch for a couple months now in production.

mahavirj, 2025-09-02:
@bryghtlabs-richard 

I took a deeper look and it appears that your proposed change should be good enough. Earlier, I was skeptical about `NGHTTP2_MAX_PAYLOADLEN` but that controls maximum frame payload size. Please send a reminder to your upstream PR and let's hope that it gets merged.

closed 2025-10-08 by mahavirj